softwarecrash/Solar2MQTT-ESP8266 #206: New working device

so4t, 2025-01-16:
### Please enter the name of the device manufacturer

SUMRY

### Please enter the device type

MPS-VIII ECO 6.2KW

so4t, 2025-01-16:
[image]

MotherBoard report this is
Device_Model = VMII-NXPW5KW
Protocol_ID = P130

###

**Photo MotherBoard**

[image]

**Photo CommunicationBoard**

[image]

so4t, 2025-01-16:
Maybe this question for another issue, But...

How to marge this project and [Pylontech-Battery-Monitoring](https://github.com/irekzielinski/Pylontech-Battery-Monitoring) ?

My inverter have 2 interface for connection WiFi (RS232) and BMS (RS485). I connected battery to BMS interface and chose protocol PYLON RS485. Wemos D1 mini with Solar2MQTT connected by WiFi (RS232) interface.

I think it's possible use one Wemos D1 mini receive DATA from 2 device

softwarecrash, 2025-01-17:
> Maybe this question for another issue, But...
> 
> How to marge this project and [Pylontech-Battery-Monitoring](https://github.com/irekzielinski/Pylontech-Battery-Monitoring) ?
> 
> My inverter have 2 interface for connection WiFi (RS232) and BMS (RS485). I connected battery to BMS interface and chose protocol PYLON RS485. Wemos D1 mini with Solar2MQTT connected by WiFi (RS232) interface. Battery manufacture by CERRNSS with BMS KS48100
> 
> I think it's possible use one Wemos D1 mini receive DATA from 2 device

i think thats not easy possible

closed 2025-01-17 by softwarecrash