softwarecrash/Solar2MQTT-ESP8266 #219: [HELP] Stoped working

wesleyff100, 2025-03-04:
### A clear and concise description of what the problem is.

I Was using normaly, then sudentlly stoped working, i tryed flashing again, reconfigure all, but still dont work.

Websocket LOG:

WebSerial] Connected...
[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Try Autodetect Protocol

[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

----------------- End Autodetect -----------------

Bootcount:0 Reboot reason:4

ProzessData P:0 C:0

WS data send

Mosquito MQTT LOG:

2025-03-04 16:47:48: New connection from 192.168.100.156:64983 on port 1883.
2025-03-04 16:47:48: New client connected from 192.168.100.156:64983 as Solar2MQTT-33CDDC (p2, c1, k15, u'admin').
2025-03-04 16:47:58: New connection from 172.30.32.2:57600 on port 1883.
2025-03-04 16:47:58: Client <unknown> closed its connection.
2025-03-04 16:49:58: New connection from 172.30.32.2:53676 on port 1883.
2025-03-04 16:49:58: Client <unknown> closed its connection.
2025-03-04 16:50:44: Client Solar2MQTT-33CDDC has exceeded timeout, disconnecting.

MQTT Explorer:
[image]

Configuration: 

[image]

Inverter: Sumry MPS ECO 3.6k 110v

softwarecrash, 2025-03-04:
just for  test try ther latest pre release, but it sound  that your communication port on the  device or the max232 chip is broken

wesleyff100, 2025-03-17:
Another RS232 adapter, new ESP8266, appears to be functioning properly now after testing. Is it normal the ESP to get kind hot to touch? i am using 1 Second MQTT Refresh (sec)

softwarecrash, 2025-03-17:
1sec mqtt refresh is to low, use 3 secon minimum.
the esp can get hot arround 40-50° but not so hot that it cant touch, so when it get so hot you have issue with power supply or bad esp

wesleyff100, 2025-03-17:
I will increase the MQTT refresh time and check how I can calculate the consumption in kWh within Home Assistant and check the power supply of the ESP. Thank you!

closed 2025-03-17 by wesleyff100