softwarecrash/Solar2MQTT-ESP8266 #222: New working device

Olexandr-Burdukovskii, 2025-03-14:
### Please enter the name of the device manufacturer

Must

### Please enter the device type

PH1900 EXP

Olexandr-Burdukovskii, 2025-03-14:
[image]

IMPORTANT! That's a USB port for the Wifi dongle. It's located on the side and has "Wifi" label. It has regular USB power pins, but data pins are instead are RS485 bus, so we can power the scheme and get the data from a single port.

Olexandr-Burdukovskii, 2025-03-14:
My implementation looks like this, and it's supplying the data to HA already. Just need to print a case
[image]

set-st, 2025-03-18:
[image]
I`m using a little bit different converter. what is a pin D5 to do? I own pv/ph18 series inverter.

Olexandr-Burdukovskii, 2025-03-18:
> [image] I`m using a little bit different converter. what is a pin D5 to do? I own pv/ph18 series inverter.

So i've been using the scheme for pv18 from another similar ticket. This one is different only in that it uses USB as both power and data source. 

Here is the original scheme:
https://github.com/softwarecrash/Solar2MQTT/issues/210

softwarecrash, 2025-03-18:
thank you for the great report

closed 2025-03-18 by softwarecrash