softwarecrash/Solar2MQTT-ESP8266 #226: 2400baud rate for MUST PV18-3024

ElectroYarik, 2025-03-20:
### Is your feature request related to a problem? Please describe

Works great with pv18-3024, maybe with others, if rebuild the project with "Serial.begin(2400);"

### Describe the solution you'd like

Тeed a choice between 2400 and 9600 baud, in the menu or automatically if the connection fails

### Describe alternatives you've considered

Manually change baud rate and rebuild project

ElectroYarik, 2025-03-24:
> wich software version you use? default the autodetect work with 2400

Solar2MQTT_esp01_1m_1.1.8_ESP01.bin

ElectroYarik, 2025-03-25:
I apologize, I got confused in the version. I downloaded the bin file of one version, and edited another :) The branch can be deleted, and I will figure out how github works

softwarecrash, 2025-03-25:
No problem

closed 2025-03-25 by softwarecrash