softwarecrash/Solar2MQTT-ESP8266 #227: [HELP] <title>POW-HVM6.2K-PRO no Protocol detection

jfsScience, 2025-03-21:
### A clear and concise description of what the problem is.

I can't connect to my to POW-HVM6.2K-PRO via RS232:

[image]

Kind regards
Jens

softwarecrash, 2025-03-24:
please provide pictures of your wireing, softwareversion ...

jfsScience, 2025-03-27:
[image]

[image]

[image]

softwarecrash, 2025-03-27:
Can you confirm that the rs485 Module Work? Most of this little Things are crap.
And get the actual pre Release

jfsScience, 2025-03-27:
Thanks a lot! I changed the RS232 TTL Adapter and now it works!

[image]

softwarecrash, 2025-03-27:
Great

closed 2025-03-27 by softwarecrash