softwarecrash/Solar2MQTT-ESP8266 #228: [HELP]  How to download ver. 2.0.0-Pre6 sourcecode&

ElectroYarik, 2025-03-25:
### A clear and concise description of what the problem is.

I get version 1.2.0-Pre4A6 instead of version  2.0.0-Pre6

softwarecrash, 2025-03-26:
its the wrong linking from github, just grab the main source

softwarecrash, 2025-03-27:
ah my fault, the actual work is under the modbus-rework branch, its 6A6 and similar to the latest relase and full working

softwarecrash, 2025-04-01:
> Good day. Please tell me the firmware version of solar 2mqtt in which data from 2 pv solar panels is visible. I have an ANERN 10200 inverter. It has 2 PV inputs for solar panels. Thank you.

the is no official firmware with this

softwarecrash, 2025-04-01:
> does an unofficial one exist?

no

softwarecrash, 2025-04-29:
updated the minify script, use this from the modbus rework.

but i have stoped working on rework, the esp doenst have enugh ram, so the crashes and other strange behavior comes from this.

ange007, 2025-04-29:
@softwarecrash 
So, for now, there's no point in taking the existing branch as a basis and trying to make changes there?

By the way, something periodically prevented me from even uploading the update on the new version (maybe the serial monitor is acting up like crazy). Maybe it would be good to make it possible to turn it off, and add a pause in case of errors.

softwarecrash, 2025-04-29:
> @softwarecrash
> So, for now, there's no point in taking the existing branch as a basis and trying to make changes there?
>
> By the way, something periodically prevented me from even uploading the [...]

You can try it, but the serial behavior is Most when the ESP Crashes when run Out of RAM, have dine serveral Tests but dint find a way at the Moment to free Up nore RAM.

closed 2025-06-13 by github-actions[bot]