softwarecrash/Solar2MQTT-ESP8266 #229: [HELP] <title>

Megakniga, 2025-04-07:
### A clear and concise description of what the problem is.

device works with wifi net exept inverter. maybe it is protocol connection or something else?

### Additional context

Dear friend
i am ukranian guy and very exited about your job especially project about Solar MQTT DIY inverter connection to Home Assistant.
I also like make things by my hands but not so familiar with any software.
So i made all what you wrote in GitHub. I have an Anenji inverter brand model ANJ-HHS-11KW-48V-WIFI.
it has its own datalogger but its not connect to HA.
I made your cool device and it works with wifi net exept inverter. maybe it is protocol connection or something else.  
could you help me out?

webconcole gave me respond this:
[WebSerial] Connecting...
[WebSerial] Connected...
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
Try Autodetect Modbus device
Init MUST PV/PH 18 protocol, baud 19200, modbusAddr 4
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Init DEYE protocol, baud 9600, modbusAddr 1
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
[WebSerial] Ping timeout.
[WebSerial] Connecting...
[WebSerial] Connected...
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Init Anenji protocol, baud 9600, modbusAddr 1
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
----------------- End Autodetect -----------------
Bootcount:0 Reboot reason:4
----------------- Start Autodetect -----------------
Try Autodetect Protocol
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
Try Autodetect Protocol
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
Try Autodetect Modbus device
Init MUST PV/PH 18 protocol, baud 19200, modbusAddr 4
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Init DEYE protocol, baud 9600, modbusAddr 1
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Init Anenji protocol, baud 9600, modbusAddr 1
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
----------------- End Autodetect -----------------
ProzessData P:0 C:0
No connection to MQTT Server: 
----------------- Start Autodetect -----------------
Try Autodetect Protocol
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
Try Autodetect Protocol
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
Try Autodetect Protocol
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
Sending [2] to Device

[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
Try Autodetect Modbus device
Init MUST PV/PH 18 protocol, baud 19200, modbusAddr 4
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Init DEYE protocol, baud 9600, modbusAddr 1
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
[WebSerial] Ping timeout.
[WebSerial] Connecting...
[WebSerial] Connected...
Response Timed Out
Time-out
Request failed!
Init Anenji protocol, baud 9600, modbusAddr 1
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
[WebSerial] Ping timeout.
[WebSerial] Connecting...
[WebSerial] Connected...
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!

softwarecrash, 2025-04-09:
i remember that the anji devices talk over modbus, so they need another modul for it, but i dont know the protocol fot this device, its possible that is not supported

Megakniga, 2025-04-10:
Thank you. I 've write letter to sell company for tech connection about our problem. So, have to wait a little.

cutik, 2025-04-17:
Don't want to create another issue, but I have the same with 2 different Anenji inverters (4 and 6.2 kW). It just cycles over different Modbus protocols without success. After all, I ordered a pre-assembled kit from @softwarecrash. Hopefully, it will arrive in Ukraine shortly after Easter so that I can provide more details/insights. There is an implementation for Home Assistant to retrieve data from the default Chinese logger [here](https://github.com/sabatex/NetDaemonApps.InverterAnenji-4kw-7.2kw); however, this is not an option for me, as the HA installation is remote from the inverter - I would like to send data to the MQTT broker directly over the internet

softwarecrash, 2025-04-17:
> Don't want to create another issue, but I have the same with 2 different Anenji inverters (4 and 6.2 kW). It just cycles over different Modbus protocols without success. After all, I ordered a [...]

the PCb are not from me, @all-solutions make this. so there is a problem, when your device talsk over modbus, the PCB doenst work, they have actual only RS232

cutik, 2025-04-17:
Oh, I thought I can support your work that way :(  I am a bit lost with all that RS232/485 Modbus stuff, totally new to me. Just want to make that stuff working and not sure what are my next steps. I tried pre pre-release build, which basically should support Anenji modbus (I guess it's related to that one - https://github.com/softwarecrash/Solar2MQTT/issues/172). Should I try connecting device to RS485 instead? Again - please excuse me for dumb questions :)

softwarecrash, 2025-04-17:
the basic problem is that modbus need too much ram, so the esp running out of it, i have a crude fix to prevent crashes when memory run out, but actual modbus is a beta try to implement. i dont know how to implement a real workung solution without getting some limitations, ant all-sollutions need to create a new pcb with both chips on it, these are all on try&testing
only the confirmed (and maybe some similar) devices that stand on the confirmed device list will sure work with the PCB

cutik, 2025-04-17:
Ok, sounds a bit complicated. Anyway, if there is something I can help with - I can do that, for instance assembling device on ESP32 instead of Wemos D1 Mini and be your guinea pig :)

cutik, 2025-04-17:
got it, thanks for the replies!

majeka, 2025-05-20:
Hello!
Please tell me if there are any results on this problem?
I also can't connect my inverter and I'm really waiting for a solution.

PavelGordovoy, 2025-06-17:
Hello!
I have an inverter Anenji 6.2KW  and I can't connect it either :(
I would really like to have a connection
we are waiting for new releases and fixes
Thank you for your work!

PavelGordovoy, 2025-06-17:
> there is no progress for anji devices, i dont get some protocols or registers at the moment

Can I help? I have an inverter nearby and I am always online.
If I can help, I can run tests, provide remote access, or follow your instructions.
I have the necessary equipment and programming knowledge.

softwarecrash, 2025-06-17:
> > there is no progress for anji devices, i dont get some protocols or registers at the moment
> 
> Can I help? I have an inverter nearby and I am always online. If I can help, I can run tests, provide remote access, or follow your instructions. I have the necessary equipment and programming knowledge.

its a bit tricky, we need to find out the registers and what is what, you can try it like bruteforce with a modbus register scanner code (something on the internet) but the hard part is to find out wich register reflect what of datatype and value

PavelGordovoy, 2025-06-19:
https://github.com/syssi/esphome-smg-ii?tab=readme-ov-file
I found this project.
I managed to connect to my inverter, but it works unstable. I don't know why.
But this project has a description of the protocols and registers

PavelGordovoy, 2025-06-21:
Received documentation from Anenji

https://docs.google.com/document/d/1nEVLuzQhgJR8osoApvlmsg-BIbu-fkYl/edit?usp=drive_link&ouid=103918979794088033013&rtpof=true&sd=true

majeka, 2025-07-23:
> for archiv here as pdf. [ANENJI RS232.docx.pdf](https://github.com/user-attachments/files/20845836/ANENJI.RS232.docx.pdf)
> 
> looks good, but at the moment i dont have much time to check and implement it, when you have programming skills you can check it to build in

Hello!
Did you manage to fix something while working with Anenji?

closed 2025-09-08 by github-actions[bot]