softwarecrash/Solar2MQTT-ESP8266 #230: [HELP] <title>Have you test it with Easun SMH-II-3.2k

jlmontes1697, 2025-04-14:
### A clear and concise description of what the problem is.

i have follow all instructions to make the solar2mqtt.
i get upload the the code and make the "RS232 TO TTL" and upload to esp8266 changing the pins 
if i run whitout inversor, the MCU connect to MQTT. I connect the inversor get "Stack trace"
0:57:19.242 > <<<stack<<<
20:57:19.242 > 
20:57:19.242 > 0x4021aec7 in ModbusMaster::ModbusMasterTransaction(unsigned char) at ??:?
20:57:19.242 > 0x402307fc in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
20:57:19.242 > 0x4022c090 in _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c:246
20:57:19.242 > 0x4022bf61 in _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c:196 (discriminator 1)
20:57:19.242 > 0x402307fc in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
20:57:19.242 > 0x402309f9 in _svfprintf_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:528
20:57:19.242 > 0x402307fc in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
20:57:19.242 > 0x401037db in rcReachRetryLimit at ??:?
20:57:19.242 > 0x402309f9 in _svfprintf_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:528
20:57:19.242 > 0x4020e0c1 in MODBUS::getModbusValue(unsigned short, modbus_entity_t, unsigned short*) at ??:?
20:57:19.242 > 0x4020e413 in _ZN6MODBUS24readModbusRegisterToJsonEPK17modbus_register_tPN11ArduinoJson8V741PB2210JsonObjectE$part$0 at modbus.cpp:?
20:57:19.242 > 0x402172ac in WebSerialClass::_send(unsigned char const*, unsigned int) at ??:?
20:57:19.242 > 0x40210000 in WiFiClient::connect(IPAddress, unsigned short) at ??:?
20:57:19.242 > 0x40217348 in WebSerialClass::write(unsigned char const*, unsigned int) at ??:?
20:57:19.242 > 0x4020e725 in MODBUS::parseModbusToJson(modbus_register_info_t&, bool) at ??:?
20:57:19.242 > 0x4020eb0c in MODBUS::retrieveModel() at ??:?
20:57:19.242 > 0x40209b14 in writeLog(char const*, ...) at ??:?
20:57:19.242 > 0x40209b14 in writeLog(char const*, ...) at ??:?
20:57:19.242 > 0x4020ec1a in MODBUS::autoDetect() at ??:?
20:57:19.242 > 0x4020df38 in MODBUS::Init() at ??:?
20:57:19.242 > 0x402023f5 in PI_Serial::autoDetect() at ??:?
20:57:19.242 > 0x40228ef4 in std::_Function_handler<void (AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int), WebSerialClass::begin(AsyncWebServer*, char const*)::{lambda(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)#2}>::_M_manager(std::_Any_data&, std::_Function_handler<void (AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int), WebSerialClass::begin(AsyncWebServer*, char const*)::{lambda(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)#2}> const&, std::_Manager_operation) at WebSerial.cpp:?
20:57:19.242 > 0x40202440 in PI_Serial::Init() at ??:?
20:57:19.242 > 0x40229ef8 in DhcpServer::isRunning() const at ??:?
20:57:19.242 > 0x4020a586 in setup at ??:?
20:57:19.242 > 0x40229ef8 in DhcpServer::isRunning() const at ??:?
20:57:19.242 > 0x40229ef8 in DhcpServer::isRunning() const at ??:?
20:57:19.242 > 0x40229ef8 in DhcpServer::isRunning() const at ??:?
20:57:19.242 > 0x40229ef8 in DhcpServer::isRunning() const at ??:?
20:57:19.242 > 0x40229ef8 in DhcpServer::isRunning() const at ??:?
20:57:19.242 > 0x40229ef8 in DhcpServer::isRunning() const at ??:?
20:57:19.242 > 0x40229ef8 in DhcpServer::isRunning() const at ??:?
20:57:19.242 > 0x40229ef8 in DhcpServer::isRunning() const at ??:?
20:57:19.242 > 0x4020a5e0 in saveConfigCallback() at ??:?
20:57:19.242 > 0x40228318 in std::_Function_handler<void (), void (*)()>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
20:57:19.242 > 0x40228310 in std::_Function_handler<void (), void (*)()>::_M_invoke(std::_Any_data const&) at ??:?
20:57:19.242 > 0x4020cdd4 in prozessData() at ??:?
20:57:19.242 > 0x4022846c in std::_Function_handler<void (), bool (*)()>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
20:57:19.242 > 0x40228464 in std::_Function_handler<void (), bool (*)()>::_M_invoke(std::_Any_data const&) at ??:?
20:57:19.242 > 0x4020ce8c in onEvent(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at ??:?
20:57:19.242 > 0x40228430 in std::_Function_handler<void (AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int), void (*)(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
20:57:19.242 > 0x40228410 in std::_Function_handler<void (AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int), void (*)(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)>::_M_invoke(std::_Any_data const&, AsyncWebSocket*&&, AsyncWebSocketClient*&&, AwsEventType&&, void*&&, unsigned char*&&, unsigned int&&) at ??:?
20:57:19.242 > 0x4020ce8c in onEvent(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at ??:?
20:57:19.242 > 0x402283ec in std::_Function_handler<void (AsyncWebServerRequest*), setup::{lambda(AsyncWebServerRequest*)#10}>::_M_manager(std::_Any_data&, std::_Function_handler<void (AsyncWebServerRequest*), setup::{lambda(AsyncWebServerRequest*)#10}> const&, std::_Manager_operation) at main.cpp:?
20:57:19.242 > 0x40208e30 in std::_Function_handler<void (AsyncWebServerRequest*), setup::{lambda(AsyncWebServerRequest*)#10}>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) at main.cpp:?
20:57:19.242 > 0x40224184 in loop_wrapper() at core_esp8266_main.cpp:?
20:57:19.242 > 0x40100805 in cont_wrapper at ??:?
20:57:19.242 > 
20:57:19.242 > 
20:57:19.242 > --------------- CUT HERE FOR EXCEPTION DECODER ---------------

### Additional context

Can you help me with the problem. Tnk

softwarecrash, 2025-04-16:
Looks Like ancokmonnissue with the memory, wich Version you are trying?
You can try the Code unser the modbus-rework branch,nähere is a try to handle Insuffizient memory

jlmontes1697, 2025-04-16:
I tried that version, and it doesn't give the error, but I still have the same problem. The inverter isn't returning any data. The inverter is an Easun SMH-II-3.2K (supposedly Voltronic compatible).
I've tried changing the position of the TX/RX cables on the MCU (ESP8266) and on the RS232-TTL, and nothing is returned.
I've changed the program to test. In the "String PI_Serial::requestData(String command)" function, I added:
writeLog("----------Send:%s--Response:%s--Length:%i",command.c_str(), commandBuffer.substring(0,1),commandBuffer.length());
and this is what it returns to me:
18:22:42.169 > Try Autodetect Protocol - Baud:2400
18:22:42.699 > ----------Send:QID--Response:--Length:0
18:22:43.226 > ----------Send:QPI--Response:--Length:0
18:22:43.792 > ----------Send:^P005PI--Response:₆₅₅₃₃ --Length:1
18:22:43.798 > Try Autodetect Protocol - Baud:2400
18:22:44.331 > ----------Send:QID--Response:₆₅₅₃₃ --Length:1
18:22:44.858 > ----------Send:QPI--Response:--Length:0
18:22:45.416 > ----------Send:^P005PI--Response:₆₅₅₃₃ --Length:1
18:22:45.422 > Try Autodetect Protocol - Baud:2400
18:22:45.947 > ----------Send:QID--Response:--Length:0
18:22:46.474 > ----------Send:QPI--Response:--Length:0
18:22:47.033 > ----------Send:^P005PI--Response:₆₅₅₃₃ --Length:1

softwarecrash, 2025-04-16:
you need the protocol sheet of the device to know what it talks

jlmontes1697, 2025-04-16:
It's what I need, but I can't find it.
The only thing I can find is that it uses RS232, and in some places, RS585.