softwarecrash/Solar2MQTT-ESP8266 #235: [HELP] Victor NM-ECO-6.2KW Plus

46eh01, 2025-05-10:
### A clear and concise description of what the problem is.

I have a Victor NM-ECO-6.2KW Plus.
It connects with older Versions of Watchpower (in my case 1.04) - with later Versions (with ModBus support) it won't connect.
Version 1.04 of Watchpower doesn't have a "Debug" field in the context menu.
Is there any chance to use it with Solar2MQTT ?

### Additional context

[image]
[image]

46eh01, 2025-05-10:
I'm very sorry - i used the wrong Version (2.0.0-pre). With the current stable Version (1.1.8) it works without any Problems. 
So.. Victor NM-ECO-6.2KW Plus works perfectly!

PS: Wiring on RJ-45 Port is:

1 - RXD
2 - TXD
4 - Vcc (13-15V)
8 - GND

softwarecrash, 2025-05-11:
thanks for the information.
the pre releases are unstable and incompatible with some devices actual, there is a coding stop until solve some problems