softwarecrash/Solar2MQTT-ESP8266 #242: [HELP]  Support for InfiniSolar V4 TWIN 6.2KW

umarjamilpc, 2025-06-11:
### A clear and concise description of what the problem is.

Hello, I wanted to know does it work with this model (InfiniSolar V4 TWIN 6.2KW).

Inverter Link: https://maxpower.com.pk/product/suntronic-pv-7000-pro/

[image]

softwarecrash, 2025-06-12:
https://github.com/softwarecrash/Solar2MQTT/wiki/Confirmed-Working-Device-List

DastardlyBaker, 2025-06-24:
Yes it does. The model is listed under Voltronics Power in the Working Device Section

umarjamilpc, 2025-06-27:
Hello, First of all i thank you for supporting this project it help me alot to integrate this into my home assistant. Is there any way to get data with 1ms response. My d1-mini freeze when set to less than 10 sec. also tell me can we replicate this project to esp32. or can you share binary for esp32.

DastardlyBaker, 2025-06-27:
> Hello, First of all i thank you for supporting this project it help me alot to integrate this into my home assistant. Is there any way to get data with 1ms response. My d1-mini freeze when set to [...]

Also, the inverter you are using only updates those value every 10 second. No point setting the acquisition rate to anything lower.

closed 2025-08-12 by github-actions[bot]